nasa/cFS-GroundSystem #12: Resolve build warnings

skliper, 2019-07-12:
2 warnings from cc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609

/home/jhageman/cFS/cFS-dev/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c: In function ‘ProcessStringArgument’:
/home/jhageman/cFS/cFS-dev/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c:239:16: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘long int’ [-Wformat=]
         printf("String Length is %d.\n", stringLength);
                ^

/home/jhageman/cFS/cFS-dev/tools/cFS-GroundSystem/Subsystems/cmdUtil/sendUdp.c: In function ‘SendUdp’:
/home/jhageman/cFS/cFS-dev/tools/cFS-GroundSystem/Subsystems/cmdUtil/sendUdp.c:74:12: warning: implicit declaration of function ‘atoi’ [-Wimplicit-function-declaration]
     port = atoi(portNum);
            ^

mrosburg, 2019-07-26:
Okay, I try this issue.

skliper, 2019-07-26:
Heads up on the first one, preference is to fix the format vs cast.

jphickey, 2019-07-26:
On printf statements it is generally not possible to fix the format when the type is a fixed width int (i.e. uint32).  A cast is generally needed here. My recommendation is to use %ld conversion AND cast the argument to "long int" as well.

skliper, 2019-07-26:
stringLength is not a fixed point int.  It's defined as long, just use %ld.

jphickey, 2019-07-26:
Yes I concur then, if it is actually "long" then no cast.

skliper, 2019-08-14:
Any updates on this issue?  Do you have an expected completion date?

mrosburg, 2019-08-14:
> Assigned #12 <https://github.com/nasa/cFS-GroundSystem/issues/12> to
> @mrosburg <https://github.com/mrosburg>.

Not sure if you'll get this but I have been delayed in doing this ticket
because of higher priority work.

Mike

mrosburg, 2019-08-27:
> @mrosburg <https://github.com/mrosburg> - do you foresee addressing this
> one soon? If not planning reassignment to @avan989
> <https://github.com/avan989>. We need the build warnings resolved for the
> end of summer release.

You can reassign it.

avan989, 2019-08-29:
I can do this.

PR #26 (2019-09-16): Integration Candidate 20190906
**Describe the contribution**
Merge of CCB 20190904 approved commits.  
Fixes #24, fixes #4, fixes #12, fixes #6, fixes #11, fixes #5, fixes #7, fixes #14, fixes #1 

**Testing performed**
Steps taken to test the contribution:
1. Checked out http://github.com/nasa/cFS master and followed setup instructions
2. Checked out ic-20190906
3. Followed standard build instructions (no cmdUtil build warnings observed)
4. Started core-cpu1 in build/exe/cpu1 directory
4. Ran make in tools/cFS-GroundSystem/Subsystem/cmdUtil to make cmdUtil executable
5. Started ground too in  [... cut]
commit 96a0bdee (2019-09-16): Fix #12, Resolve build warnings
PR #45 (2019-12-19): Fix #44, Release Prep
**Describe the contribution**
Fix #12
 - Updated README
 - removed custom license document
 - added standard Apache 2.0
 - Updated copyright release version cFE 6.6 -> 6.7

**Testing performed**
1. Standard build, unit test and execute

**Expected behavior changes**
 - No impact to behavior

**System(s) tested on:**
 - Hardware: cFS Dev Server 2
 - OS: Ubuntu 18.04
 - Versions: cFE 6.7.0 related versions and OSAL 5.0.0

**Contributor Info**
Jacob Hageman - NASA/GSFC
closed 2019-09-16 by skliper