nasa/cFS-GroundSystem #46: Recommended lgtm issues

avan989, 2019-12-20:
**Is your feature request related to a problem? Please describe.**
Recommendation from lgtm:

CommandParser.py
```
line 20:

import sys
--
  | Import of 'sys' is not used.
21 | import csv
  | Import of 'csv' is not used.
22 | import getopt
  | Import of 'getopt' is not used.
23 | import subprocess
  | Import of 'subprocess' is not used.
24 | import shlex
  | Import of 'shlex' is not used.
```

HTMLDocsParser.py
```
line 20:

import sys
--
  | Import of 'sys' is not used.
21 | import csv
  | Import of 'csv' is not used.
22 | import getopt
  | Import of 'getopt' is not used.
23 | import subprocess
  | Import of 'subprocess' is not used.
24 | import shlex
  | Import of 'shlex' is not used.

```

GroundSystem.py
```
line 26:
import socket
--
  | Import of 'socket' is not used.
27 | import zmq
  | Import of 'zmq' is not used.
28 |  
29 | from PyQt4 import QtGui, QtNetwork, QtCore
  | Import of 'QtCore' is not used.Import of 'QtNetwork' is not used.
```

RoutingService.py
```
line 23:
import sys
--
  | Import of 'sys' is not used.
24 | import os
  | Import of 'os' is not used.
25 | import socket
26 | import zmq
27 |  
28 | from PyQt4 import QtGui, QtNetwork, QtCore
  | Import of 'QtGui' is not used.Import of 'QtNetwork' is not used.
```

Paramete.py
```
import csv
--
  | Import of 'csv' is not used.

from PyQt4 import QtGui, QtNetwork
--
  | Import of 'QtNetwork' is not used.
29 | from ParameterDialog import Ui_Dialog
30 | from HTMLParser import HTMLParser
  | Import of 'HTMLParser' is not used.

line 104:
idx = 0
--
  | The global variable 'idx' is not used.

line 111:
params = ''
--
  | The global variable 'params' is not used.

line 125:
idx = int(arg) # comand index in command definition file
--
  | The global variable 'idx' is not used.
```

UdpCommands.py
```
import csv
--
  | Import of 'csv' is not used.

from HTMLParser import HTMLParser
--
  | Import of 'HTMLParser' is not used.
```

CHeaderParser.py
```
import sys, os
--
  | Import of 'os' is not used.

line 437:
typeNew = parser.findKeyword(dataTypesOrig[-1])
--
  | The global variable 'typeNew' is not used.
```

GenericTelemetry.py
```
import pdb
--
  | Import of 'pdb' is not used.
```

TlmUDPSender.py
```
import sys
--
  | Import of 'sys' is not used.
```

EventMessage.py
```
line 151 - 153
udpPort  = 10000
--
  | The global variable 'udpPort' is not used.
152 | appId = 999
153 | tlmDefFile = "not-needed.txt"
  | The global variable 'tlmDefFile' is not used.

line 171:
udpPort = arg
--
  | The global variable 'udpPort' is not used.

line 175:
tlmDefFile = arg
--
  | The global variable 'tlmDefFile' is not used.

line 193 - 195:
py_endian = '<'
--
  | The global variable 'py_endian' is not used.

py_endian = '>'
--
  | The global variable 'py_endian' is not used.
```

GenericTelemetry.py
```
line 166:
udpPort  = 10000
--
  | The global variable 'udpPort' is not used.

line 186:
udpPort = arg
--
  | The global variable 'udpPort' is not used.

line 252:
for j in range (i, 40):
--
  | The global variable 'j' is not used.
```

UdpCommands.py
```
line 247:
for i in range(len(cmdDesc)):
--
  | The global variable 'i' is not used.

line 249:
for i in range (len(cmdDesc), 26):
--
  | The global variable 'i' is not used.
```

CommandSystem.py
```
for j in range (i, 22):
--
  | The global variable 'j' is not used.

```

TelemetrySystem.py
```
line 82:
appIdString = appIdString + "%02X" % ord(packetData[1])
--
  | The value assigned to local variable 'appIdString' is never used.

line 167:
def dumpPacket(packetData):
--
  | The value assigned to local variable 'dumpPacket' is never used.

line 192:
send_host = "127.0.0.1"
--
  | The value assigned to local variable 'send_host' is never used.

193 | send_port =  tlmPagePort[i]
  | The value assigned to local variable 'send_port' is never used.
```

**Requester Info**
Anh Van, NASA Goddard

lbleier-GSFC, 2020-04-13:
This issue has been updated to be Python-related only; C issues are consolidated in issue #43

skliper, 2020-06-24:
@lbleier-GSFC have these been resolved as part of your previous work?  Doesn't look like these are listing the autogenerated files,

skliper, 2020-06-24:
@lbleier-GSFC see https://lgtm.com/projects/g/nasa/cFS/alerts/?mode=list&lang=python for the latest set of issues.  These all look to be against hand generated files.  Can they be resolved?

lbleier-GSFC, 2020-06-24:
> @lbleier-GSFC see https://lgtm.com/projects/g/nasa/cFS/alerts/?mode=list&lang=python for the latest set of issues. These all look to be against hand generated files. Can they be resolved?

I'm very confused. I fixed these already. I am looking at my local copies and they are fixed. Not only that, the auto-generated ones should be deleted because I changed the filenames (added the `Ui_` prefix)

skliper, 2020-06-24:
> > @lbleier-GSFC see https://lgtm.com/projects/g/nasa/cFS/alerts/?mode=list&lang=python for the latest set of issues. These all look to be against hand generated files. Can they be resolved?
> 
> I'm very confused. I fixed these already. I am looking at my local copies and they are fixed. Not only that, the auto-generated ones here should be deleted because I changed the filenames (added the `Ui_` prefix)

Did you?  I just looked in master for the first LGTM alert, and it was still in the code (QtGui not used in CommandSystemDialog.py).  Did you submit something that didn't get checked in?  What are you looking at locally?  Is there a branch you have locally that needs a PR?

lbleier-GSFC, 2020-06-25:
> Did you? I just looked in master for the first LGTM alert, and it was still in the code (QtGui not used in CommandSystemDialog.py). Did you submit something that didn't get checked in? What are you [...]

That particular file is auto-generated. I fixed the alerts for files that were hand-written

skliper, 2020-06-25:
> > > > @lbleier-GSFC see https://lgtm.com/projects/g/nasa/cFS/alerts/?mode=list&lang=python for the latest set of issues. These all look to be against hand generated files. Can they be resolved?
> > > 
> > > 
> > > I'm very confused. I fixed these already. I am looking at my local copies and they are fixed. Not only that, the auto-generated ones here should be deleted because I changed the filenames (added the `Ui_` prefix)
> > 
> > 
> > Did you? I just looked in master for the first LGTM alert, and it was still in the code (QtGui not used in CommandSystemDialog.py). Did you submit something that didn't get checked in? What are you looking at locally? Is there a branch you have locally that needs a PR?
> 
> That particular file is auto-generated. I fixed the alerts for files that were hand-written.
> 
> I am looking at the branch I made for this issue

Copy.  Looks like LGTM hasn't run an updated analysis on the latest master yet.  I haven't been able to trigger it, but apparently it does a daily pull.

lbleier-GSFC, 2020-06-25:
> Copy. Looks like LGTM hasn't run an updated analysis on the latest master yet. I haven't been able to trigger it, but apparently it does a daily pull.

What's odd is that I renamed auto-generated files such as these to have the `Ui_` prefix in their filename. The pre-renamed files should be deleted as far as git is concerned. I am not sure why they are still coming up

skliper, 2020-06-25:
@lbleier-GSFC coming up where?  In LGTM it is because the last run was on a previous commit.  You can see the commit it used and compare it with master (it's using the 6/10 IC merge last I checked).

commit aa15da9d (2020-07-01): Merge pull request #96 from lbleier-GSFC/fix46lgtmissues

Fix #46 Recommended lgtm python issues
commit f52c69d1 (2020-07-01): Fix #46 Recommended lgtm python issues

Problems shown in this issue were fixed in #72 as part of
updates/refactoring. Other lgtm issues addressed here.
Problems in auto-generated .py files not addressed.
Auto-generated .py files renamed with Ui_ prefix.
The lgtm.yml file must be updated to exclude these
See nasa/cFS#92
PR #109 (2020-07-26): Integration Candidate: 2020-06-24
**Describe the contribution**
Fix #46 
Resolve #103 
Resolve #112 

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/108/checks

**Expected behavior changes**

PR #96 - Fixes more lgtm warnings

PR #108 - Allows users to change the byte offsets for sending commands and parsing telemetry, to support different header versions or other implementations of cFS

PR #113 - Adds a file to store version information and reports version upon ground-system startup.

PR #114 - Fixes mismatch between GUI and backend leading to AttributeError bug (Added in  [... cut]
closed 2020-07-26 by astrogeco