nasa/cFS-GroundSystem #57: Scrub/Redo GroundSystem command

avan989, 2020-03-13:
**Is your feature request related to a problem? Please describe.**
Some of the commands are outdated and does not match current cFS. 

**Describe the solution you'd like**
Redo all command to ensure everything is updated. 

**Describe alternatives you've considered**
Scrubbing and fixing the parameter files manually (more difficult than just redoing the command)

**Additional context**
Add-on to issue #31

**Requester Info**
Anh Van, NASA Goddard

skliper, 2020-03-13:
@dmknutsen I think you are also working this?

dmknutsen, 2020-03-17:
I am.  I submitted a similar ticket + rebuilt all of the commands and submitted a pull request yesterday to fix this (under issue #59).

skliper, 2020-03-17:
Marking as duplicate of #59 and closing

closed 2020-03-17 by skliper