nasa/cFS-GroundSystem #61: cFS ground system GUI Python code should generate cmd UDP directly

CDKnightNASA, 2020-03-17:
**Is your feature request related to a problem? Please describe.**
Currently the ground system code calls cmdUtil to send commands to the "spacecraft" via UDP messages to CI. This is clunky (and a giant security issue, not that we're that concerned about that...) and Python's quite capable of generating CCSDS UDP packets directly, skipping the need to call cmdUtil.

**Describe the solution you'd like**
I suggest implementing a library function to send commands, and that library function can directly generate CCSDS command UDP traffic. cmdUtil could even be re-written to be a Python program that just wraps that library function rather than requiring a compiler to build an executable version of cmdUtil.

**Describe alternatives you've considered**
Could just stick with current architecture. Currently this requires compiling cmdUtil and ensuring you're in the right location relative to cmdUtil for it to be called. Current calls assume the executable is in "../cmdUtil/cmdUtil" (so, wherever you start the GUI, it assumes cmdUtil is in a subdirectory called "cmdUtil" under the parent directory.

**Additional context**
Add any other context about the feature request here.

**Requester Info**
Christopher.D.Knight@nasa.gov

skliper, 2020-03-17:
Good point, the complexity/dependency is unnecessary.

Guess at the core needs:
 - a command line capability to send commands (supports automation, currently used in CI and I've seen it used in I&T)
 - a simple lab tool for building/sending commands and receiving/displaying telemetry in human readable format (typically used during early development)

Not to limit the upper end if someone wants to add the capabilities (scripting, graphics, triggered actions, etc), but I think the sweet spot is between nothing and a full blown ground system.

jphickey, 2020-03-17:
Once again, I'll just remind folks that this is (mostly) implemented in the EDS library, which is now apache 2/open source and as such can be integrated if/when the desire exists.  (https://github.com/jphickey/cfe-eds-framework)

The EDS runtime library has Python bindings that allow one to build CMD packets using EDS packet definitions, so one can use this to generate a packet.

lbleier-GSFC, 2020-06-16:
@CDKnightNASA it would seem that I need to at least partially implement this fix in order to implement #103

skliper, 2020-07-27:
Resolved by #103.

PR #108 (2020-07-06): Resolves #103 Custom header mechanism
**Describe the contribution**
Fixes #103, implemented the ability to allow users to select the header version byte offset, or custom offsets. In order to support this, some relatively major changes had to be implemented:

- A Python-native "mini" `cmdUtil` to send UDP data. The GroundSystem app in its current form is no longer dependent on the external C `cmdUtil` app (also fixes #61, at least partially)
- Using Python's `mmap` module to share data between processes

**Testing performed**
Steps taken to test the contribution:
1. Make changes to code
2. Run [... cut]
closed 2020-07-27 by skliper