nasa/cFS-GroundSystem #79: IP address dropdown not fully working

lbleier-GSFC, 2020-04-29:
**Describe the bug**
The dropdown in the main window indicates "All" or "127.0.0.1" as options. When "All" is selected, it seems to work fine, but a Python exception is thrown when "127.0.0.1" is selected:

```
Traceback (most recent call last):
  File "GroundSystem.py", line 87, in startTlmSystem
    subscription = '--sub=GroundSystem.' + selectedSpacecraft + '.TelemetryPackets'
TypeError: must be str, not bytes
```
**To Reproduce**
Steps to reproduce the behavior:
1. Run the Python app
2. Select "127.0.0.1" from the IP address dropdown
3. Try opening the telemetry page

**Expected behavior**
Telemetry page should open and no crashes should occur

**Reporter Info**
Leor Bleier NASA GSFC\Code 582 (with support from @skliper)

lbleier-GSFC, 2020-05-07:
The crash no longer occurs thanks to #72 but having the menu on `127.0.0.1` doesn't seem to work at all, not sure why

thnkslprpt, 2023-03-22:
Working fine for me at the moment on Debian. Whatever was causing this may have been fixed up in one of the previous updates.  

[image]