nasa/cFS-GroundSystem #93: Pointer alignmnet issues on 32 bit ARM

Jkillelea, 2020-05-22:
It looks like some struct has changed size and it no longer aligns nicely with 32 bit ARM. It's possible that this software simply shouldn't be compiled on this platform and instead it should be reserved for an x86/x86_64 based ground station.

Steps to reproduce the behavior:
1. Clone cFS from https://github.com/nasa/cFS 
2. Compile

Expected behavior:
Structs sizes should align so that casts can be made on platforms that do not allow unaligned access (such as ARM).

Code snips:
```
Scanning dependencies of target mission-all
Scanning dependencies of target cmdUtil
[ 14%] Building C object tools/cFS-GroundSystem/Subsystems/cmdUtil/CMakeFiles/cmdUtil.dir/cmdUtil.c.o
[ 28%] Building C object tools/cFS-GroundSystem/Subsystems/cmdUtil/CMakeFiles/cmdUtil.dir/SendUdp.c.o
/home/pi/upstream-cfs/tools/cFS-GroundSystem/Subsystems/cmdUtil/SendUdp.c: In function ‘SendUdp’:
/home/pi/upstream-cfs/tools/cFS-GroundSystem/Subsystems/cmdUtil/SendUdp.c:90:20: error: cast increases required alignment of target type [-Werror=cast-align]
         inet_ntoa(((struct sockaddr_in*)result->ai_addr)->sin_addr), port);
                    ^
cc1: all warnings being treated as errors
tools/cFS-GroundSystem/Subsystems/cmdUtil/CMakeFiles/cmdUtil.dir/build.make:86: recipe for target 'tools/cFS-GroundSystem/Subsystems/cmdUtil/CMakeFiles/cmdUtil.dir/SendUdp.c.o' failed
make[7]: *** [tools/cFS-GroundSystem/Subsystems/cmdUtil/CMakeFiles/cmdUtil.dir/SendUdp.c.o] Error 1
CMakeFiles/Makefile2:462: recipe for target 'tools/cFS-GroundSystem/Subsystems/cmdUtil/CMakeFiles/cmdUtil.dir/all' failed
make[6]: *** [tools/cFS-GroundSystem/Subsystems/cmdUtil/CMakeFiles/cmdUtil.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make[5]: *** [all] Error 2
CMakeFiles/mission-all.dir/build.make:57: recipe for target 'CMakeFiles/mission-all' failed
make[4]: *** [CMakeFiles/mission-all] Error 2
CMakeFiles/Makefile2:164: recipe for target 'CMakeFiles/mission-all.dir/all' failed
make[3]: *** [CMakeFiles/mission-all.dir/all] Error 2
CMakeFiles/Makefile2:171: recipe for target 'CMakeFiles/mission-all.dir/rule' failed
make[2]: *** [CMakeFiles/mission-all.dir/rule] Error 2
Makefile:212: recipe for target 'mission-all' failed
make[1]: *** [mission-all] Error 2
Makefile:120: recipe for target 'all' failed
make: *** [all] Error 2
pi@raspberrypi:~/upstream-cfs $ uname -a
Linux raspberrypi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux
```
System observed on:
 - Raspberry Pi Zero
 - OS: Linux 4.19
 - cFS git commit `a4345d87c01b1818bdb217ffd4dcce8bf86e33fe`, cFS-GroundSystem git commit `1c31f94f3b63d3ff9c5da35ca9f8825a89ef3258`

Jkillelea, 2020-05-22:
On closer inspection, the offending code seems to be a `printf`, casting from `struct addrinfo` to `struct sockaddr_in`. 

https://github.com/nasa/cFS-GroundSystem/blob/master/Subsystems/cmdUtil/SendUdp.c#L89

Weird. Both types appear to have the same alignment but different sizes. 

Demo code:

```
pi@raspberrypi:~ $ cat aligntest.c 
#include <stdio.h>
#include <stdalign.h>
#include <sys/types.h>
#include <sys/socket.h>
#include <netdb.h>

int main() {
	printf("sizeof addrinfo: %lu\n", sizeof(struct addrinfo));
	printf("alignof addrinfo: %lu\n", alignof(struct addrinfo));
	printf("sizeof sockaddr_in: %lu\n", sizeof(struct sockaddr_in));
	printf("alignof sockaddr_in: %lu\n", alignof(struct sockaddr_in));
	return 0;
}
pi@raspberrypi:~ $ gcc aligntest.c -o aligntest
pi@raspberrypi:~ $ ./aligntest 
sizeof addrinfo: 32
alignof addrinfo: 4
sizeof sockaddr_in: 16
alignof sockaddr_in: 4
```

jphickey, 2020-05-22:
The problem is that it's casting a `struct sockaddr*` to a `struct sockaddr_in *` ... which is generally the way its supposed to be done except that `struct sockaddr` probably is what's missing the alignment.

Roundabout fix would be to create a union with both a `struct sockaddr` and `struct sockaddr_in` member, then copy the data..  The code should also really be checking `ai_addrlen` and `ai_family` if not already doing so.

skliper, 2020-05-22:
Or use getnameinfo?

Jkillelea, 2020-05-23:
In the meantime, it looks like `arch_build_custom.cmake` can be edited to suppress pointer alignment errors. Probably not a good long term fix.

skliper, 2020-08-31:
Also observed by @hukuzatuna. Recreated on gcc 8+ with cast-align=strict.

PR #130 (2020-09-03): Fix #93, Refactor and resolve cast align warning
**Describe the contribution**
Fix #93 - replaces old code that caused a cast-align warning when strict

 - refactored and removed unnecessary code while also following recommended model for getaddrinfo.
 - removed old windows support/defines/etc (likely not tested for years, no longer supported).

**Testing performed**
Built, sent successful command to cFE

**Expected behavior changes**
No more warning, less fragile.

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: bundle main + this commit

**Additional cont [... cut]
commit 99439463 (2020-09-03): Fix #93, Refactor and resolve cast align warning
commit bafd03b2 (2020-09-03): Merge pull request #130 from skliper/fix93-strict-align

Fix #93, Refactor and resolve cast align warning
PR #133 (2020-09-09): Integration Candidate: 2020-09-02
**Describe the contribution**
Fix #93 
Fix #128 

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/136/checks

**Expected behavior changes**
PR #130 - Replaces old code that caused a cast-align warning when strict.  Refactored and removed unnecessary code while also following recommended model for `getaddrinfo`.  Removed old windows support/defines/etc (likely not tested for years, no longer supported).

PR #129 - Reduce the size of the `strncpy` so that it ensures there's a null byte at the end of the string buffer.

**System(s) tested on**
 [... cut]
closed 2020-09-09 by yammajamma