nasa/cFS-GroundSystem #97: FileNotFound ERROR when trying to send commands using GroundStation.py GUI

PaulaBosca, 2020-05-28:
When I try to send any command using the GUI I receive the following python error: 

FileNotFoundError: [Errno 2] No such file or directory: 'home/user/cFS/tools/cFS-GroundSystem/Subsystems/cmdGui/../cmdUtil/cmdUtil' :  'home/usercFS/tools/cFS-GroundSystem/Subsystems/cmdGui/../cmdUtil/cmdUtil '

I have successfully installed all the necessary packages as described in the Guide and when the GroundSystem.py is started and I click the telemetry system or command system buttons I get the appropriate 
'Attempting to wait for UDP messages 
Telemetry System started. Subscribed to Groundsystem
Command System started'

Only when I try to send any command i get the Traceback error. I have both cmdGui and cmdUtil directories under Subsystems with the same files as shown on the Github. 

I'm running cfs on Ubuntu 18.04 through a VM. I haven't started editing any files or developing any applications, just wanted to get everything set and and working before I did. The whole system was downloaded yesterday so all build versions are the most recent.

skliper, 2020-05-28:
Looks like the error message has changed, but try:

https://github.com/nasa/cFS-GroundSystem/blob/5c14375741732382cf0b635073b0b3aab25cdca8/Guide-GroundSystem.txt#L96-L118

skliper, 2020-05-28:
Probably would help if this step was in the initial setup...

PaulaBosca, 2020-05-28:
Perfect, I somehow missed that! Problem fixed, thank you!

closed 2020-05-28 by PaulaBosca