nasa/cFS-GroundSystem #99: Fix Markdown for Ground System Guide

astrogeco, 2020-05-29:
**Is your feature request related to a problem? Please describe.**
Multiple numbering and formatting inconsistencies in Guide-GroundSystem.md.
Duplicate file with Guide-GroundSystem.txt

**Describe the solution you'd like**
Remove Guide-GroundSystem.txt and update Markdown file using [remark](https://github.com/remarkjs/remark), [prettier](https://prettier.io/) or [Tidy Markdown](https://github.com/notslang/tidy-markdown)

Long-term create a implement style guide for Markdown

Rename file to follow CamelCase.

**Describe alternatives you've considered**
Short manual scrub to fix numbering and code blocks.

zacnyjegomosc, 2020-05-30:
Absolutely. Can I grab this issue?

astrogeco, 2020-06-01:
> Absolutely. Can I grab this issue?

Take it away!

The markdown formatting examples are just meant as guidelines. You might want to look into the different toolchains and do a pros/cons between them. I envision something similar to what we did with clang-format. See the cFS repo's .travis file

skliper, 2020-06-01:
Looks to me like just an end of code quote got messed up...

astrogeco, 2020-06-03:
There's some weird numbering stuff happening in the steps section too.

PR #117 (2020-07-30): Fix #99, Guide format fix
**Describe the contribution**
Fix #99 - deletes old text file and fixes formatting of markdown

**Testing performed**
Viewed in GitHub.  No code changes.

**Expected behavior changes**
Displays guide as expected.

**System(s) tested on**
GitHub

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
commit 8fb7f4a7 (2020-07-30): Fix #99, Guide format fix
commit 9b052707 (2020-07-30): Merge pull request #117 from skliper/fix99-guide-format-fix

Fix #99, Guide format fix
closed 2020-07-30 by astrogeco