nasa/cFS-GroundSystem #104: 2.1.11 fails to build on RHEL 7.8

johnphamngc, 2020-06-17:
**Describe the bug**
2.1.11 fails to build on RHEL 7.8, due to endianness macros, as part of cFS build, with or without devtoolset-9:

```
[ 14%] Building C object tools/cFS-GroundSystem/Subsystems/cmdUtil/CMakeFiles/cmdUtil.dir/cmdUtil.c.o
cd /tmp/cFS/build/tools/cFS-GroundSystem/Subsystems/cmdUtil && /opt/rh/devtoolset-9/root/usr/bin/cc -D_XOPEN_SOURCE=600  -g   -std=c99 -pedantic -Wall -Wstrict-prototypes -Wwrite-strings -Wpointer-arith -Wcast-align -Werror -o CMakeFiles/cmdUtil.dir/cmdUtil.c.o   -c /tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c
/tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c: In function ‘ProcessField’:
/tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c:370:13: error: implicit declaration of function ‘htobe16’ [-Werror=implicit-function-declaration]
  370 |     *orig = htobe16((be16toh(*orig) & ~mask) | (templong & mask));
      |             ^~~~~~~
/tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c:370:22: error: implicit declaration of function ‘be16toh’ [-Werror=implicit-function-declaration]
  370 |     *orig = htobe16((be16toh(*orig) & ~mask) | (templong & mask));
      |                      ^~~~~~~
/tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c: In function ‘main’:
/tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c:666:33: error: implicit declaration of function ‘htole16’ [-Werror=implicit-function-declaration]
  666 |                     tempint16 = htole16(tempint16);
      |                                 ^~~~~~~
/tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c:708:33: error: implicit declaration of function ‘htobe32’ [-Werror=implicit-function-declaration]
  708 |                     tempint32 = htobe32(tempint32);
      |                                 ^~~~~~~
/tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c:710:33: error: implicit declaration of function ‘htole32’ [-Werror=implicit-function-declaration]
  710 |                     tempint32 = htole32(tempint32);
      |                                 ^~~~~~~
/tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c:752:33: error: implicit declaration of function ‘htobe64’ [-Werror=implicit-function-declaration]
  752 |                     tempint64 = htobe64(tempint64);
      |                                 ^~~~~~~
/tmp/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/cmdUtil.c:754:33: error: implicit declaration of function ‘htole64’ [-Werror=implicit-function-declaration]
  754 |                     tempint64 = htole64(tempint64);
      |                                 ^~~~~~~
cc1: all warnings being treated as errors

```

**To Reproduce**
Build CFS GroundSystem on RHEL 7.8 as part of the cFS build. However, using the cmdUtil makefile works fine. 

**Expected behavior**
Build completes correctly as part of cFS makefile build

**System observed on:**

 - OS: RHEL 7.8
 - Versions : cFS-GroundSystem 2.1.11

**Additional context**
This is apparently fixed by changing the macro defined at the top of cmdUtil.c from `#define _BSD_SOURCE_` to `#define _BSD_SOURCE`

**Reporter Info**
John N Pham, Northrop Grumman

skliper, 2020-06-17:
Related to #95.  Trade is add the _BSD_SOURCE define, use -std=gnu99 (if this works), or re-implement the functions.

jphickey, 2020-06-17:
Note, `-std=gnu99` does _not_ work for this platform.  It is not a GNU extension, and is not enabled by gnu99.

For this to build on RHEL you currently have to add `_BSD_SOURCE` to mission_build_custom.cmake, in addition to the `_XOPEN_SOURCE` which may be already there from the example, e.g.:
```
add_definitions(
    -D_XOPEN_SOURCE=600
    -D_BSD_SOURCE
)
```

My preference would still be to implement #95 which wouldn't require this extension on some platforms.

PR #106 (2020-06-24): Fix #104, Short term build fix for implicit endian functions
**Describe the contribution**
Quick temporary fix for implicit declaration of endian functions on some systems (RH/CentOs).  Note the real fix will be #95, custom implementation of these functions and removal of the messy defines.

Fix #104 

**Testing performed**
Built cleanly on CentOS (confirmed build error before change and error gone after)

**Expected behavior changes**
No build errors on CentOS

**System(s) tested on**
 - Hardware: Intel i5, WSL
 - OS: CentOS 7.6.1810
 - Versions: bundle w/ this change

**Additional context**
 [... cut]
commit ddee8c82 (2020-06-24): Fix #104, Short term build fix for implicit endian functions (#106)
PR #107 (2020-06-25): Integration Candidate: 2020-06-17
**Describe the contribution**
Fix #98
Fix #104

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/106/checks

**Expected behavior changes**
PR #100 - Change all individual UI elements to table widgets. Update backend code accordingly

PR #106 - Temporary fix for implicit declaration of endian functions on some systems (RH/CentOs). Note the real fix will be #95, custom implementation of these functions and removal of the messy defines. No build errors on CentOS

**System(s) tested on**
Ubuntu: Bionic

**Additional context**
Part of https://gith [... cut]
closed 2020-06-25 by astrogeco