nasa/cFS-GroundSystem #123: "start" script or at least make GroundSystem.py executable

CDKnightNASA, 2020-08-20:
**Is your feature request related to a problem? Please describe.**
Currently users have to explicitly type `python3 GroundSystem.py` to start the ground system. (And if they type `bash GroundSystem.py` or `python GroundSystem.py` [and the default Python is 2] they get errors.) At minimum the GroundSystem.py file should be made executable so users can use `./GroundSystem.py` and not worry about what is the correct scripting language. This will also likely help with GUI file managers, enabling starting from the GUI without having to go to a terminal.

It might be better to have a short `start` or `go` or even just rename it to `GroundSystem` (no extension) or some other obvious command that starts the ground system. Given the folder's full of .py files, having a file that's distinct will be helpful to new users. Would like CCB's consideration on what to name such a command.

**Requester Info**
Christopher.D.Knight@nasa.gov

sunny1401, 2020-10-03:
Hi - I was wondering if I can try doing this?

CDKnightNASA, 2020-10-05:
> Hi - I was wondering if I can try doing this? - Not sure if I can contribute to this?

Absolutely!

PR #137 (2021-03-22): Fix #123, create package for cfs-groundsystem
**Describe the contribution**
A clear and concise description of what the contribution is.
- Include explicitly what issue it addresses [e.g. Fixes #123 ]

**Expected behavior changes**
A clear and concise description of how this contribution will change behavior and level of impact.
 - Behavior Change: (The user will be able to type command 'startg' and start the ground system from anywhere on the system)
 

**System(s) tested on**
 - OS: [macOS Catalina v 10.15.7]

**Additional context**
I installed the package with command pip3 install -e relati [... cut]
commit 18cbb52d (2021-03-22): Fix #123, #121, Make into package and close all windows on exit
commit 331b408d (2021-03-22): Merge pull request #137 from zachar1a/main

Fix #123, create package for cfs-groundsystem
PR #171 (2021-03-22): cFS-GroundSystem Integration candidate: 2021-03-23
## Describe the contribution
Fix #165, Updated TBL and SB tlm for an operational TLM display 
Fix #169, Add Contributing Guide 
Fix #123 and Fix #121, create package for cfs-groundsystem 

## Testing performed

See:
- <https://github.com/nasa/cFS/pull/223/checks>
- <https://github.com/nasa/cFS-GroundSystem/pull/171/checks>

## Expected behavior changes

### PR #166 

Corrects values in sb and tbl hk-tlm.txt to allow the TBL and SB tlm pages to open.

### PR #170 

Adds a contributing guide that links to the main cFS contributing guide.

### PR [... cut]
closed 2021-03-22 by astrogeco