nasa/cFS-GroundSystem #134: Enable Tlm IP address communication fails at initialization

hukuzatuna, 2020-09-04:
**Describe the bug**
With an unmodified build of integration-candidate including the fix for #852, opening cFS Ground System->cFE/CFS Subsystem Commands->Enable Tlm and entering a remote IP address in the Input field and clicking "send" does not properly initialize communication with CFS on the satellite machine. Other behaviors (see below) indicate the entered IP address is not propagated internally either.

**To Reproduce**
Steps to reproduce the behavior:

1. Configure a "satellite" machine and a remote "Ground Station" machine.
2. Start cFS/build/exe/cpu1/cpu/core-cpu1 on the satellite machine, as instructed in the README
3. On the ground station machine, click through "CFS Ground System"->"Start Command System"->"Enable Tlm"
4. Enter the IP address of the remote satellite machine in the Input field
5. Click "send"
6. Note that the terminal on the satellite machine shows no data connection or receipt of packets (this is in contrast to the behavior if you run the Ground Station system on the same machine as the satellite CFS processes)
7. Then, on the Subsystem Commands window, note the "Send to" fields all contain localhost (127.0.0.1) and cannot be changed.
8. Click "Display Page" next to any subsystem, e.g., "SB No-Op"
9. Change the "Send To" value at the top of the window to be the remote IP address of the satellite machine
10. Click "Send" next to CFE_SB_NOOP_CC
11. Note the terminal window on the satellite machine displays information about receipt of the no-op

**Expected behavior**
The Ground Station system should properly use the entered remote IP address and initialize TLM. The entered IP address should propagate throughout the subsequent Ground Station windows and child windows.

**Code snips**
I haven't figured out exactly where the problem lies, so cannot provide a context diff.

**System observed on:**
 - Raspberry Pi 4 (both computers)
 - OS: Raspberry Pi OS, latest build and patched
 - Versions [e.g. cFE 6.6, OSAL 4.2, PSP 1.3 for mcp750, any related apps]
    + Raspberry Pi OS version 10 (Buster)
    + CFS - integration-candidate branch as of 2020-09-03, 1700 hours
    + cFE v6.0.0-rc1+dev42 release v6.7.0

**Additional context**

1. Running systems in permutations of root/non-root does not change the behavior (so not a socket() permission problem)
2. Running systems on the same machine works, but not on remote machines (implies entered remote IP address not be used properly internally - perhaps a failed class attribute update - maybe in an instance of GroundSystem.ipAddressesList[]?)
3. Manually changing the "Send To" IP address in "Subsystem Commands"->[anything]->"Display Page" produces output as expected on the satellite machine (which means the UDP communications work, it's just the IP address handling inside the Ground System code has a bug?)

**Reporter Info**
Phil Moyer (hukuzatuna on GitHub, hukuzatuna@gmail.com or phil@moyer.ai)
No company (yet), individual user

skliper, 2020-09-04:
Related to #79

skliper, 2020-09-04:
Note of caution - we often break integration-candidate as it's our staging branch for CI (https://github.com/nasa/cFS/wiki/The-cFS-CCB-Process).  Main is the more stable development branch.

hukuzatuna, 2020-09-04:
Thank you for the caution. I didn't (don't) know how to merge #852 in with
main, so I checked out integration-candidate. Please let me know if you'd
like me to do anything else to help investigate!

jmtobin-uh, 2022-09-13:
@hukuzatuna 
I seem to have this same issue when running a cFS satellite on one machine and the GS on a separate machine. When cFS and the GS are on the same machine and I use localhost ip, tlm works as expected.

closed 2026-02-26 by dzbaker