nasa/cFS-GroundSystem #138: GroundSystem.py run error

GitHubForPP, 2020-10-10:
**Describe the bug**
run python3 GroundSystem.py prompt
File "GroundSystem.py", line 102
    subscription += f'.{selectedSpacecraft}.TelemetryPackets'
                                                            ^
SyntaxError: invalid syntax

**To Reproduce**
Steps to reproduce the behavior:
1.sudo apt-get install python3-pyqt5
2.sudo apt-get install python3-zmq
3.sudo apt-get install libcanberra-gtk-module
4.cd Subsystems/cmdUtil/ && make
5.python3 GroundSystem.py

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - Hardware
 - OS: UBUNTU 16.04

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Full name and company/organization if applicable

closed 2020-10-14 by GitHubForPP

skliper, 2020-10-14:
@GitHubForPP Did you get this to work?

GitHubForPP, 2020-10-14:
> @GitHubForPP您使它起作用了吗？

Yes,you can use it by updating the python version to 3.7.