nasa/cFS-GroundSystem #186: Switch to PySide2 to avoid licensing problems with PyQt5

pavll, 2021-07-27:
**Is your feature request related to a problem? Please describe.**
PyQt5 is not an official implementation of Qt in Python. The new _official_ alternative from Qt Company is https://wiki.qt.io/Qt_for_Python (PySide2).

**Describe the solution you'd like**
Switching the project's dependency on PyQt5 to PySide2.

**Describe alternatives you've considered**
Staying at PyQt5.

astrogeco, 2021-07-27:
What are some benefits of PySide besides the official support?

pavll, 2021-07-27:
- PySide comes with a license under the LGPL, meaning it is simpler to incorporate into commercial projects when compared with PyQt. 
- It allows the programmer to use QtQuick or QML to establish the user interface. 
- It allows the usage of primary widget-based user interface resources.
- There are available no official Python-specific resources on classes in PyQt5, meaning you can only use the data in C++. It can be quite confusing for a programmer that does not have extensive experience in either C++ or Qt.
- PySide represents the official set of Python bindings backed up by the Qt Company.

pavll, 2021-07-27:
*the current version is PySide6