nasa/cFS-GroundSystem #189: Jetbrains IDE, virutualenv and Pipenv support within .gitignore

pavll, 2021-07-28:
**Is your feature request related to a problem? Please describe.**
If you work with an IDE like PyCharm from Jetbrains you have to be careful when committing not to send the `.idea` folder by mistake because it is not listed in the `.gitignore`. Also missing is the exclusion of `venv` folders and `Pipenv` files in the `.gitignore`. I think some developers will be happy about this change.

**Describe the solution you'd like**
Add ... to `.gitignore:`

- .idea folder
- venv folder
- Pipenv file
- Pipenv.lock file

**Describe alternatives you've considered**
Do the excluding manually every time you have some of these in your code.

**Additional context**
I will add a PR that includes the above changes.

**Requester Info**
pavll

astrogeco, 2021-08-09:
Hey @pavll did you close this by accident?

pavll, 2021-08-14:
Yes

PR #190 (2021-09-01): Fix #189, Virtualenv and Pipenv .gitignore support
Structured .gitignore

**Describe the contribution**
- Fixes #189 

**Testing performed**
 1. made a commit with the new .gitignore

**Contributor Info - All information REQUIRED for consideration of pull request**
Paul Oberosler, Individual
commit 00005b9c (2021-09-01): Merge pull request #190 from pavll/fix-189-jetbrainside-venv-pipenv-support-gitignore

Fix #189, Virtualenv and Pipenv .gitignore support
commit 5f571bbb (2021-09-01): Fix #189, Virtualenv and Pipenv .gitignore support

Structured .gitignore
PR #195 (2021-09-01): cFS-GroundSystem Integration candidate: 2021-08-31
**Describe the contribution**

PR #190

- Fix #189, Virtualenv and Pipenv .gitignore support

PR #194

- Fix doc, comment, and message typos 

**Testing performed**

cFS-GroundSystem Checks <https://github.com/nasa/cFS-GroundSystem/pull/195/checks>
cFS Bundle Checks <https://github.com/nasa/cFS/pull/348/checks>

**Expected behavior changes**
See PRs

**System(s) tested on**
Ubuntu

**Additional context**
Part of <https://github.com/nasa/cFS/pull/348>

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of [... cut]
closed 2021-09-01 by astrogeco