nasa/cFS-GroundSystem #248: Misleading wording about ports used for telemetry.

vladvoicux64, 2025-02-18:
**Describe the bug**
Commands-Telemetry.md specifies the port at which telemetry packets are sent being UDP 1234. In practice UDP 1234 is the port that the TLM app listens to, and UDP 1235 is the default port at which telemetry is actually sent to the tlmGUI.
**To Reproduce**
Steps to reproduce the behavior:
1. Start spacecraft software
2. Start GS software and enable TLM as indicated in the README.md in the cFS repository
3. Run `ss -tulnp`, notice port 1234 being used by "core-cpu1" (i.e. spacecraft) and port 1235 being used by "python3" (ground station):
```
% ss -tulnp  
                     
Netid State   Recv-Q  Send-Q   Local Address:Port    Peer Address:Port Process                                                                
udp   UNCONN  0       0              0.0.0.0:1234         0.0.0.0:*     users:(("core-cpu1",pid=36008,fd=11))                                 
udp   UNCONN  0       0              0.0.0.0:1235         0.0.0.0:*     users:(("python3",pid=36059,fd=16))                                   
udp   UNCONN  0       0              0.0.0.0:54071        0.0.0.0:*     users:(("core-cpu1",pid=36008,fd=15))
```
4. Run  `sudo tcpdump -i lo udp`, notice spacecraft sends packets to port 1235 for TLM:
```
%  sudo tcpdump -i lo udp

16:42:20.443949 IP localhost.54071 > localhost.1235: UDP, length 40
16:42:20.944602 IP localhost.54071 > localhost.1235: UDP, length 44
16:42:20.944640 IP localhost.54071 > localhost.1235: UDP, length 300
16:42:20.944651 IP localhost.54071 > localhost.1235: UDP, length 20
16:42:20.944660 IP localhost.54071 > localhost.1235: UDP, length 160
```

**Expected behavior**
Wording in original is interpreted as "UDP port 1234 is where TLM packets are sent", where in reality that is the port that the TO_LAB listens to for commands.
**Software used**
You need to install `tcpdump` to correctly observe this behaviour.

**System observed on:**
 - M1 Ultra chipset
 - OS: Debian 12.5 ARM64
 - cFS-GroundSystem Aquilla release

**Reporter Info**
Voicu Ioan-Vladut

RR-HSz, 2025-02-21:
I would extend your topic a little bit:

According to ci_lab_interface_cfg.h: 

_"In order to allow multiple instances of CFE to run on the same host, the processor number - 1 is added to this value.  This, if this is set to "1234",  then the following ports will be used at runtime:

Processor 1: port 1234
Processor 2: port 1235
Processor 3: port 1236"_

So, the command port of CPU2 (1235) will conflict with the telemetry port of CPU1 (also 1235).

My solution for this was to increase the CI port with 2 per processor.
Maybe it should be checked by the experts.

vladvoicux64, 2025-02-21:
Good point, I can test with a second CPU and report back. I'll do it after I finish work and report back.

vladvoicux64, 2025-02-21:
I tested with a second CPU and the default setup does not even start, throwing an exception in RoutingService.py at line 59, that the address is already in use. I looked in the file and the TO_Lab listening port is hardcoded at 1235. I think it is safe to say that this issue is safe to close with my rewrite, as it is regarding the default config. I think it's best to open a new issue regarding the issue of port assigning.

commit b5a976bf (2025-02-25): Fix misleading wording in Commands-Telemetry.md

Fix #248, the original wording misleads readers to believe that the default port for receiving telemetry is 1234, in reality it being 1235, and the former being the port that the telemetry app uses to listen for commands.
PR #249 (2025-02-25): Fix misleading wording in Commands-Telemetry.md
The original wording misleads readers to believe that the default port for receiving telemetry is 1234, in reality it being 1235, and the former being the port that the telemetry app uses to listen for commands.

**Describe the contribution**
Fixes #248 by improving the wording and adding information obtained by research

**Testing performed**
Tested using `ss` and `tcpdump` to track listening ports and sent packets. Proof can be found in the addressed issue.

**Expected behavior changes**
Readers will hopefully not be mislead in the future and w [... cut]
closed 2025-02-25 by dzbaker